meshtastic/Meshtastic-Apple #810: 🐞 [Bug]: iOS and OSX TestFlight apps "forget" remote admin for all submenus except "Radio Configuration"

akohlsmith, 2024-07-21:
### Firmware Version

2.4.0.1093

### What did you do?

Normally when you go to the settings tab you select a remote node under the "configure" heading, and then you go to whichever sub-menu you want to remotely configure. The latest TestFlight apps on both OSX and iOS allow me to select the remote node, but when I go to a submenu (e.g. "User") the details are for the device I'm connected to via BLE, not the remote node. Note that the correct remote node is displayed if I look at the sub-menus under the "Radio Configuration" heading.

### Expected Behavior

Expected behaviour is for the app to remember that it's remotely configuring a node over LoRa

### Current Behavior

* Remote node is selected under "Configure"
* going into "LoRa" under "Radio Configuration" remembers I'm configuring a remote node
* going into any other submenu under the other headings displays the config for the BLE-connected node, not the remote-admin node.

I have also noticed that when I tap on an item such as "User" the UI very briefly shows that "Bluetooth" was tapped, then the screen moves and shows me the User settings (but for the BLE-connected node, not the remotely administrated node)

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-07-22:
After a reboot the app will be managing the connected node

SpudGunMan, 2024-07-22:
I put a video in discord for the issue https://discord.com/channels/867578229534359593/874835141145542716/1263767770701758475

SpudGunMan, 2024-07-23:
Rolled back to 2.3.17.1076 and issue is gone

garthvh, 2024-07-31:
I am not seeing this anymore with the latest test flight version, I have all the expected configuration now

closed 2024-07-31 by garthvh

SpudGunMan, 2024-07-31:
The same admin issue exists in build 1011 for me (not resolved)