meshtastic/Meshtastic-Apple #816: 🐞 [Bug]: Long-press options missing for nodes located in the list using a search string

clwgh, 2024-07-23:
### Firmware Version

2.3.17 (1078)

### What did you do?

Meshtastic app 2.3.17 (1078). iOS 17.5.1. iPhone SE 3rd gen.

Nodes > long press on a node > options present are:
- Add to favorites
- Hide alerts
- Exchange Positions
- Trace Route
- Delete Node

Nodes > enter a few characters to quickly locate a node > long press on node > options present are only:
- Add to favorites
- Hide alerts

In either case selecting the node and scrolling down shows the options anyway, however these used to all be available from the node list with a long press after filtering to quickly find a node.

### Expected Behavior

All options are present, regardless of whether the node was located in the list via a search string or not.

### Current Behavior

Most long press options are missing (eg Trace Route) if the node was located in the list using a search string.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Many thanks.

RPN707, 2024-07-25:
These disappeared about two weeks ago but then came back after a day or two. What you're describing to me actually sounds like the behavior I see when I'm NOT connected to a node. The offline behavior, so to speak, where you can only do local things but not mesh things.

At the same time I noticed this though, the Delete item in the node list context menu stopped working. I reported that here in issue 789 and mentioned the missing context items. That's still broken for me.

garthvh, 2024-08-01:
You are not connected to a node, only favorite and hide alert are available if you are not connected

closed 2024-08-01 by garthvh

clwgh, 2024-08-01:
You've closed this based on a skim read of RPN707's post and not the detail in my opening report.

I have 250 nodes in the node list. I want to do a traceroute to ABCD which is somewhere in the list.

I used to go to the top to "Find a node" and enter AB, which instantly reduces the list to 3 nodes, then I can long-press on ABCD and do a traceroute. Now that doesn't work because "Trace Route" doesn't appear as an option.

But if I scroll down the list and hunt for ABCD and find it, I can long-press on it and "Trace Route" does still appear as an option.

The loss of long-press options in the filtered list it a change in behaviour and I am reporting it.

clwgh, 2024-08-02:
# Node list
[image]

# Long-press on node, all options shown
[image]

# Filtered node list
[image]

# Long-press on same node, options missing
[image]

garthvh, 2024-08-02:
[image]

Current App Store version

[image]

Connected node long press options

[image]

Disconnected node long press options

garthvh, 2024-08-02:
Both of those are with a search filter.

clwgh, 2024-08-02:
Your version screenshot prompted me to check for an update and `2.4.0` is now available. It wasn't available when I created this bug report. In `2.4.0` I am seeing the same four options regardless of whether the node is located in the full list or a filtered list, so the bug is fixed in `2.4.0` and it is once again possible to traceroute to someone else's node via a long-press in a filtered list.

What do you mean by connected and disconnected nodes?

I am referring to someone else's node in the node list. My screenshots show the difference in long-press options between not using, and using, a search filter to find the SAME node in the list. Whatever you are referring to above, my relationship to the node does not change between not filtering and filtering, so you are demonstrating something else.

In `2.3.16` I would filter to find a node quickly and then long-press to traceroute. I became aware of this bug in `2.3.17` because filtering then a long-press no longer showed traceroute. I am pleased consistent UI behaviour has been restored in `2.4.0`.