meshtastic/Meshtastic-Apple #818: 🐞 [Bug]: iOS 18 - no channel messages

rinchen, 2024-07-25:
### Firmware Version

2.3.6

### What did you do?

Upgraded to public beta of iOS 18 and the channel message area is blank. 
[image]

### Expected Behavior

The ability to see channel messages like in iOS 17th

### Current Behavior

White screen

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-07-25:
There is no iOS 18 build at all yet, please report any bugs you find with the beta to apple.

Lolagatorade, 2024-07-27:
Yeah would be a pretty critical fix considering iOS 18 release is right around the corner.

Lolagatorade, 2024-07-29:
Direct messages work no issue running in latest iOS 18 dev beta. [image]
[image]

garthvh, 2024-07-29:
Keep reporting bugs to Apple

djh1997, 2024-07-31:
> Direct messages work no issue running in latest iOS 18 dev beta. [...]

Definitely seems like an app issue as opposed to an ios issue

Lolagatorade, 2024-07-31:
> Definitely seems like an app issue as opposed to an ios issue

Exactly what I was thinking but if that's where the line is drawn and the conclusion that is made, I'm just gonna stay quiet I guess. 

Because it makes no sense for it to work in one version of iOS and not in the other it just seems like API update that broke the app and the application needs to be updated to adapt to that 

Apple just makes the platform it's up to the developers to accommodate to any updates and changes.

garthvh, 2024-07-31:
Fetched properties are not working on iOS18, pretty clearly an OS issue.

Lolagatorade, 2024-07-31:
> Fetched properties are not working on iOS18, pretty clearly an OS issue.

Is that specific to ios18?
Because on ipados[image]18 it's working just fine now with my testing.
Seems to be a GUI/UX issue

Look at the end of the day you're the maintainer and you're the expert on this issue. I'm just saying that it seems like it's more than just an operating system issue given that has already been 4 beta updates and it is still ongoing. 

Attached you will see two different messages being sent from an iPad to the long fast channel successfully on two different Meshtastic devices. It is also received on my iPhone, but I am unable to open and reply. Nothing shows. But it still shows up on the little notification badge.

Maybe it's too zoomed in where the messages aren't seen because I do notice. There are some strange scaling issues on the iPad who knows just a suggestio [image]suggestion.

[image]
[image]
 [image][image]n[image]

djh1997, 2024-08-01:
looks like its being fixed in [this pull request](https://github.com/meshtastic/Meshtastic-Apple/pull/825)

djh1997, 2024-08-01:
For me some times when I use the landscape layout I can get it to load.

Lolagatorade, 2024-08-01:
> For me some times when I use the landscape layout I can get it to load.

Yeah there's something going on with aspect ratios and scaling because even in iPad when using different window modes it either works or doesent 
[image]

[image]

garthvh, 2024-08-01:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/825

garthvh, 2024-08-01:
I keep closing these issues because we have known from the start what was broken.