meshtastic/Meshtastic-Apple #826: 🐞 [Bug]: Offline .mbtiles maps in legacy mode no longer working

Kollensperger, 2024-08-01:
### Firmware Version

2.4.0 (1106)

### What did you do?

I'm on testflight and tested this with both recent testflight versions and the version on the app store.
Delete app, fresh install. 
Select "use legacy mesh map" in app settings.
Select Mesh Map in maps, select "offline" in "i" menu, Enable Offline Maps and Enable MB Tiles selected
Open mbtiles file in Files menu (also tried sharing with meshtastic)

The files I am trying are about 30, 300 MB and 1 GB and were confirmed working in prior versions (unfortunately I did not note which one.....)

### Expected Behavior

Expect the mbtiles map to display correctly

### Current Behavior

Loaded map does not display correctly, base map still remains

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

This is one of the best features with meshtastic especially for people who use the maps for more than just a hike on a path and who need customised maps.
Its unfortunate that this is slated to be dropped going forward - I hope we can have one last working version or get some help identifying which one is the final working version

garthvh, 2024-08-01:
These features are pretty much abandoned and will likely  be removed with 16 support soon

garthvh, 2024-08-01:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/829

closed 2024-08-01 by garthvh

waiotahi, 2024-11-20:
I'm putting my hand up as an ios meshtastic user who finds the new apple offline maps as unusable. Whoever thought that offline users would find the current version acceptable obviously has not used it offline. We are often in area's that apple maps has no comprehensive coverage. Please bring back the ability to use mbtiles so we can load our own maps. One ios mapping app that have excellent offline support is https://github.com/MerginMaps/mobile - see how they've enabled it