meshtastic/Meshtastic-Apple #827: 🚀 [Feature Request]: add support for apple translation api

djh1997, 2024-08-01:
### OS

iOS

### Description

Investigate the addition of support for the [apple language translation api](https://developer.apple.com/documentation/Translation) like in the [ice cubes mastodon app ](https://github.com/Dimillian/IceCubesApp) into the messages as in Europe their are lots of different langue's spoken within range.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-01:
This is already in place, feel free to use Xcode for translation

djh1997, 2024-08-01:
sorry I didn't mean to change the language of the app I meant to translate incoming messages to the same language as the app.

72A12F4E, 2024-08-01:
This is an interesting idea!

72A12F4E, 2024-08-01:
To implement this, we would just need to add a row in app settings to toggle the behavior on and off. 

Then, we would add a cult to this translations API wherever the message text is rendered. 

We could always fall back to the existing message. Text if the API fails, or add a tap action to the message bubble to allow the user to swap between the translated version or not

djh1997, 2024-08-01:
ice cubes (open source ios app on github) puts a translate button on any text that is detected to be in a foreign language it might be possible to just put the sf symbol for translate into the message bubble. 
I'll try to get a screen shot for you later if you want.
thanks for making such a great app.

djh1997, 2024-08-01:
possible relevant [pull request](https://github.com/Dimillian/IceCubesApp/pull/2065) from ice cubes

hayschan, 2025-01-11:
Good idea.

pqhf5kd, 2025-05-24:
Would love to see this. Even just the translate option when long pressing a message.