meshtastic/Meshtastic-Apple #840: 🚀 [Feature Request]: Improve readability for SNR and RSSI numbers

clwgh, 2024-08-03:
### OS

iOS

### Description

Latest versions of iOS and app as of 3rd August.

The **SNR** and **RSSI** are very difficult to read, especially on a smaller SE display and near impossible when they are coloured yellow on a white background.

Would it be possible to have these displayed as plain black text and in a slightly larger font please, and just have the signal icon above it show the colour? These numbers are used for analysing signals and at the moment I am making screenshots and zooming to see the values.

## Example
Save this screenshot on your computer desktop, display it and scale the view so that the image is **4.1 inches tall**. That shows how it looks on the SE.

[image]

Thanks

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

If you're okay changing these data to black and can tell me the file in which to make the change, I will submit a PR.

garthvh, 2024-08-03:
You can download the data as a CSV to get black text, the colors have meaning.

closed 2024-08-03 by garthvh

garthvh, 2024-08-03:
I did bump the font back up to caption from caption2 and gave removed some padding.

garthvh, 2024-08-03:
[image]

This is where the RSSI number is useful, to identify a noisy environment, this is the larger font for RSSI and smaller for the sir which has more length with the extra digits.

clwgh, 2024-08-03:
That's great, thankyou. Do you have an example with a yellow reading please? I'll scale it here and see how it looks and report back. I know the colours convey "at a glance" information; it's just that yellow text on a white background is so hard to see. I appreciate the tweak, the larger font should make quite a difference.

I'm not sure how to download a csv but I doubt it would be practical for just checking nodes as they come in; it's quicker to screenshot, open the screenshot and zoom in on the text to read it, then abort the screenshot.

clwgh, 2024-08-04:
Looks like it will be large enough to see when yellow, based on a side by side scaled to actual size (height 4.1"). Thankyou for the tweak.

[image]