meshtastic/Meshtastic-Apple #846: 🐞 [Bug]: Time synchronization in messaging and statuses (online/offline)

patrikcze, 2024-08-06:
### Firmware Version

2.4.0.46d7b82

### What did you do?

On Devices without Time sync or wrong time (RAK, T-ECHO). Messages received to device are sorted incorrectly. Message sent as a reply is sorted before message which was sent first. I guess this TIME SYNC issue is not problem only with messages, but also with ONLINE statuses etc. 

### Expected Behavior

Message received later than first message should be sorted at the end of the chat.

### Current Behavior

Message received as a reply to previous message are being sorted in front of this message in some cases at the top of chat (when time is completely unavailable on LoRa Device).

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

There seems to be strange behaviour in sending and receiving messages. Those are incorrectly sorted out, and I believe also Online Status is improperly showed due to wrong time. 

This one occurred at the beginning of the chat
[image]

These messages popped up before the message which was sent to device which automatically replied to those.
[image]

I'm not programmer. But would it be possible to consider different way of handling time in Client? I do not know if that is possible or not.

garthvh, 2024-08-06:
Time is handled entirely by the firmware, there is not any time sync in meshtastic, adding extra client logic just adds more confusion.

closed 2024-08-06 by garthvh

patrikcze, 2024-08-21:
Hi, just one question, since this has been reported by multiple people. Also on reddit. Messages are ordered improperly when time is messed on any device (which are most of them). Interestingly this does not affect Android app, just iOS according to my info. when time is synced, everything works ok. But some devices do not have GPS or properly synced time. Is there any way how to setup manually time? AT+ command? :-) just asking.

garthvh, 2024-08-21:
The android app uses the time when the messages are pulled into the app and not RX time, so it will be better in some cases and wrong in others (like any time you reconnect to a device that has received messages).  The time is managed entirely by the firmware and iOS shows the time provided by the device.