meshtastic/Meshtastic-Apple #850: MQTT does not send to everyone subscribed.

Lolagatorade, 2024-08-07:
### Firmware Version

2.4.1

### What did you do?

Recently, I got MQTT working, which was surprisingly buggy, and even with the same exact configurations would refuse to send as now (functional), even though it acknowledges every message.

Now that messages actually get through to a recipient the main issue is that not everybody subscribed to the topic can actually view the messages sent

At times I am able to see something another user may not. Or vice versa. Take note that the root topic is not very large in terms of active users.

### Expected Behavior

If a message is sent, it is expected to be received by everybody that is currently viewing and subscribe to the topic. As in how MQTT should function with topics ect. 

### Current Behavior

A message sent is not guaranteed to be seen by everybody subscribe to a root topic. Attached is an example where I reply to something somebody said, but one user is not on the same page and was confused because he saw my reply but not the context
[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

Lolagatorade, 2024-08-07:
Another example of a user replying to a message that was never received...

[image]

garthvh, 2024-08-09:
On the default long fast channel the public MQTT server is zero hopped since July 15, these messages are no longer forwarded via lora.

closed 2024-08-09 by garthvh