meshtastic/Meshtastic-Apple #860: 🐞 [Bug]: Battery percentage inconsistent

purge98, 2024-08-10:
### Firmware Version

2.4.1

### What did you do?

Checked Device Metrics Log and it shows different battery percentages for the same voltage. 

### Expected Behavior

A particular voltage should always show the same percentage.

### Current Behavior

App shows different battery percentages for the same voltage

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

[image]
[image]
[image]

closed 2024-08-10 by garthvh

purge98, 2024-08-10:
Thanks for closing it BUT why does Volts 3.51 show 18% and 30%? Doesn’t correlate.

garthvh, 2024-08-10:
There is already an open issue, please check before opening another one