meshtastic/Meshtastic-Apple #861: 🐞 [Bug]: Frequency Override

Camousse88, 2024-08-10:
### Firmware Version

2.4.1

### What did you do?

On iOS, when setting a frequency with four digits after the decimal point, such as 869.4625 (Gaulix network in France), the application rounds to three digits after the decimal, 869.463.
This results in a frequency offset.

### Expected Behavior

Get four digits after the decimal point.
[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

Camousse88, 2024-08-10:
[image]
[image]

garthvh, 2024-08-10:
There is just a float value be passed around in the app, nothing is being modified converted or rounded at all by the app.

Camousse88, 2024-08-10:
It’s not a bug it’s a feature. 🙂

More seriously. It’s not clear; rounding to 4 decimals would be more readable.

Too bad

garthvh, 2024-08-10:
There is a number formatter on there, I will see if maybe that has a default decimal precision

closed 2024-08-10 by garthvh