meshtastic/Meshtastic-Apple #862: 🐞 [Bug]: Messages show a blank screen after navigating awa and coming back

spydercs, 2024-08-10:
### Firmware Version

2.4.1

### What did you do?

While viewing a channel chat or a direct message chat, navigate away to one of the other parts of the app such as bluetooth, press the 'messages' button again to go back to the chat.

### Expected Behavior

When returning to the chat screen the original chat is displayed.

### Current Behavior

While viewing a channel chat or a direct message chat, if I navigate away to one of the other parts of the app such as bluetooth, when I come back the screen is blank. In order to see the chat again I have to back out twice and then select either 'channels' or 'direct messages' and then the channel or contact I was in so I can view the chat again.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

on an iphone 13 mini iOS version 17.5.1

Lolagatorade, 2024-08-14:
Yes, this has been an issue for me too, and it would be very nice to get it fixed iPhone 13 mini running iOS 18

I've been wanting to make a request for it, but haven't had the time so just been dealing with annoyance

YLSnewb, 2024-08-15:
Also seeing this on iPhone Pro Max 15 with iOS 17.5.1.

mosh1, 2024-08-19:
Also seeing this one. Effects usability quite a bit. iPhone Pro Max 15 and iPhone Pro 14 with iOS 17.5.1.

nbgibson, 2024-08-19:
Confirming on a 13 Pro and iOS 17.6.1. The issue can be mitigated via restarting the app but that'll have to do until it's fixed. I suspect that a wider UX pass might be needed, but that's easier said than done.

tekstrand, 2024-08-21:
@garthvh is this one on your radar? I know you've been focused on the new PKI features so this one may have slipped past. This bug really breaks some core functionality and requires a lot of extra taps to get around. I tried looking at the code but got hopelessly lost on this.

ponzano, 2024-08-21:
Can confirm it happens on iOS 18 beta 7 too

Lolagatorade, 2024-08-21:
> Can confirm it happens on iOS 18 beta 7 too

Yeah it's not a OS specific issue.