meshtastic/Meshtastic-Apple #872: 🚀 [Feature Request]: Set Node Icon Color

akrabu, 2024-08-14:
### OS

iOS, macOS

### Description

This may already be implemented, but I noticed that the background for the node icon / short name has a color, and that color persists between apps and devices.  But what I can't figure out for the life of me... is where to change that color haha

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

akrabu, 2024-08-14:
Ah, I just figured out that the color is set from your node number converted to a 32 bit integer with the last 6 hex digits used for the color.  That's good to know.  I'd still like to request it be user-configurable, though!

RCGV1, 2024-08-14:
This would require either making the user be easily able to change their user id (which would mess up a lot of things) or create a new value that determines the color.

RCGV1, 2024-08-14:
Stick to collecting all the colors on your nodes like Pokémon, it's more fun this way

closed 2024-08-14 by RCGV1

akrabu, 2024-08-14:
I was afraid of that, but it makes sense.  Thanks!