meshtastic/Meshtastic-Apple #873: 🐞 [Bug]: Messages are showing out of order

brianrowe, 2024-08-14:
### Firmware Version

2.4.1

### What did you do?

Sending and receiving messages via the IOS app

### Expected Behavior

When sending and receiving messages, they should alternate one after the other to indicate the flow of the conversation. Much like messages work in the IOS Messages app when sending an iMessage or SMS.

### Current Behavior

The app is "grouping" the messages. The messages on the left side of the screen where the responses to the messages sent on the Right side of the screen. `Msg 1` was sent and then `Res 1` was received. `Msg 2` was sent and then `Res 2` was received.

This is happening on Direct Messages and Channels

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Running IOS 17.5.1
Meshtastic IOS app 2.4.1
Wisblock Node running Firmware 2.4.1

garthvh, 2024-08-14:
You can check the timestamps by long pressing on the messages and you will see they are in the order they were received on the device.  The messages order by the RX time (or the time from the phone if the. RX time is 0)

The client app has no knowledge of what order a message was sent in.

closed 2024-08-14 by garthvh

brianrowe, 2024-08-14:
> You can check the timestamps by long pressing on the messages and you will see they are in the order they were received on the device. The messages order by the RX time (or the time from the phone if the. RX time is 0)
> 
> The client app has no knowledge of what order a message was sent in.

Here is what the same messages on the Android app look like. "Conversation" style like text messaging on your phone. From Social media posts, it sounds like others are experiencing this same issue with the IOS app.

[image]

brianrowe, 2024-08-14:
@garthvh Please see the example above. This issue should not be closed. This is a NEW issue since the previous version.

brianrowe, 2024-08-14:
@garthvh Not sure what determines the time stamps, but the Ack Time is happening before the Message time. In this example I sent the message `Test 5:27` as soon as the time on my phone switched to `5:27`. As you can see in the attached screen shot the Message Time correlates with when I sent the message, but the Ack time is before I sent the message.

Is the IOS app setting the Ack time when it receives the Acknowledgement, or is that time sent from the first Node that acknowledged it?

[image]

brianrowe, 2024-08-14:
@garthvh Is this PR from a few days ago, addressing this issue, it just has not made it to the App Store yet? https://github.com/meshtastic/Meshtastic-Apple/pull/864

fluxstorm0, 2024-08-20:
Awesome to see a fix for this. Thank you guys for the hard work! It is exciting to see this project grow.