meshtastic/Meshtastic-Apple #874: Sensor graph missing time stamp after reboot

Lolagatorade, 2024-08-16:
### Firmware Version

2.4.3

### What did you do?

After reboot it starts populating datapoints. When you look at the sensor data, there is no graph. The timestamps are missing for sometime until they populate. Eventually it works but it takes too long. The poll rate for the sensor is set to 1 minute.  

### Expected Behavior

Tike stamp should populate after a reboot basically instantly so the graph can graph over time 
[image]

### Current Behavior

As mentioned the graph does not populate because of missing time stamps 

attached is an example of 223x my node and an example of functional node. 

[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-20:
Time is managed entirely by the firmware, the app show the time it was given

closed 2024-08-20 by garthvh

Lolagatorade, 2024-08-20:
> Time is managed entirely by the firmware, the app show the time it was given

My bad forgot to close it. They did a hot fix for the firmware a day or two ago.