meshtastic/Meshtastic-Apple #875: IAQ colored dots off center when looking as column

Lolagatorade, 2024-08-16:
### Firmware Version

2.4.3

### What did you do?

Title explains it 

### Expected Behavior

Dots should line up 

### Current Behavior

Dots don’t 
[image]
[image] up 

### Participation

- [ ] I am willing to submit a pull request for this issue.

Lolagatorade, 2024-08-16:
Placing them to the left of the IAQ might solve the issue it's probably just the numbers being slightly smaller or larger and shifting it over

garthvh, 2024-08-17:
The grids are being replaced for all the logs

closed 2024-08-17 by garthvh