meshtastic/Meshtastic-Apple #878: Display Fahrenheit

Lolagatorade, 2024-08-20:
### OS

iOS, iPadOS, macOS

### Description

Currently, the application only displayed Celsius in the environmental sensor section in the settings you can enable display Fahrenheit, but this only works on the actual firmware of the device and it’s built in display. The actual application still only shows Celsius.

it would be logical for the setting to transfer not only to the device itself, but the actual application

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-20:
This setting is for the device display, measurements are all managed by your phone region

closed 2024-08-20 by garthvh

Lolagatorade, 2024-08-20:
> This setting is for the device display, measurements are all managed by your phone region

That makes sense, but sometimes it is not the intended behavior as an example. 

My phone's region is set to the United States, but for measurements and units I have it set to metric because of the nature of its ease of calculations, etc.

This is due to the reason of increased granularity when using Fahrenheit compared to Celsius, whereas I still use meters for calculating distance due to how easy it is to use so it is a weird mix of units in this use case scenario the system making assumptions actually works against the user

garthvh, 2024-08-20:
You have total control via language and settings