meshtastic/Meshtastic-Apple #884: Favorites not sticking

Lolagatorade, 2024-08-21:
### Firmware Version

2.5.0

### What did you do?

Favorites not sticking after favoriting 

### Expected Behavior

Favorite nodes should stick 

### Current Behavior

After some time it will forget favorite nodes 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-21:
Favorites require firmware 2.3.2 or greater and are stored in the node db I see favorite staying through multiple bluetooth connections and hundreds on seen nodes on MQTT, you should be able to favorite up to 99 nodes

Lolagatorade, 2024-08-21:
> Favorites require firmware 2.3.2 or greater and are stored in the node db I see favorite staying through multiple bluetooth connections and hundreds of seen nodes on MQTT, you should be able to favorite up to 99 nodes

Yes but it is not working for the newest firmware release and meshtastic update released today. On firmware 2.5.0

garthvh, 2024-08-21:
Your node db is lost when you update, this is expected.

Lolagatorade, 2024-08-21:
> Your node db is lost when you update, this is expected.

Yes, this is fine and I've experienced it before but what I mean is making new favorites they do not stick 

So let's say you add a new user as a favorite a few minutes later it will disappear in the nodes list and on the private message list