meshtastic/Meshtastic-Apple #889: Signal not populating on node list

Lolagatorade, 2024-08-21:
### Firmware Version

2.5.0

### What did you do?

Similar to the GPS issue these signal values are actually available on the device itself. Other devices running the same firmware will show it, but for some reason, my device nor my friends device shows it over the mesh network 

it is erratic because my own node through Bluetooth won’t show it, but my friends will show his signal strength yet another device with the same exact hardware that I have in my same household will show the signal strength over the mesh network…

attached will be some screenshots where you can see my friend in map view can see his own location and his signal value, but on my device running the same exact operating system and firmware with everything up-to-date it is not visible

One strange thing to keep in mind is the actual signal value that does show up on his device is very suspicious as it shows zero values 

Yet for my other device named plutonia, it does show a value that seems to be correct. This just comes to show the current state has a high variance on how it acts from device to device, even with the same firmware and hardware.

[image]
[image]
yet
[image]

[image]

### Expected Behavior

The expected behavior is on the view for everything to function in terms of populating the GPS and populating the signal values correctly and accurately.

### Current Behavior

Currently from the looks of it, the signal values can be erratic in terms of showing up zero value or failed to show up completely or show up seemingly correctly from an exact hardware and firmware devices meshed. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-21:
The signal meter only shows on nodes you are directly connected to.

closed 2024-08-21 by garthvh

Lolagatorade, 2024-08-21:
> The signal meter only shows on nodes you are directly connected to.

Yes I am directly connected through Bluetooth to agroprom. And it won't show its values. 

Yet it will show it for plutonia which is on the mesh and not directly connected...

Lolagatorade, 2024-08-21:
[image]

After a reboot and reconnect it did appear. But the node list has been a bit iffy on bug hunt

Talie5in, 2024-08-22:
> Yes I am directly connected through Bluetooth to agroprom. And it won't show its values.
> 
> Yet it will show it for plutonia which is on the mesh and not directly connected...

I think the definition of "directly connected" is not what you are expecting here.

You wont get a signal value for "your device" you are connected to over Bluetooth, but you will for a Direct "RF/LoRa" Radio that your device can see, but not one that is 1 hop away over "RF/LoRa"

JohnathonMohr, 2024-08-23:
> but you will for a Direct "RF/LoRa" Radio that your device can see, but not one that is 1 hop away over "RF/LoRa"

For what it's worth, this is not the behavior I experience. I might not get a signal bar on the node list, but I always get signal strength indicators (with varying values) in the node info screen, even if the node is multiple hops away.

Lolagatorade, 2024-08-23:
> For what it's worth, this is not the behavior I experience. I might not get a signal bar on the node list, but I always get signal strength indicators (with varying values) in the node info screen, [...]

Yeah same here but maybe that is due to MQTT being enabled?

garthvh, 2024-08-23:
The signal strength meter is only for directly connected nodes on both the list and details, you can not measure the signal from yourself to yourself so the Bluetooth  connected node does not have a signal meter either.

JohnathonMohr, 2024-08-23:
@garthvh I don't disagree with you, but at the same time I do see it for nodes I'm not directly connected to. Maybe that is its own bug 🤷‍♂️. Here's an example from the mesh map tab, where you can clearly see a node is multiple hops away, but still has a strength indicator.

[image]

The "graph" appears in the node details tab too, but that page doesn't show hops away, so not as clearly demonstrated.

Lolagatorade, 2024-08-23:
> @garthvh I don't disagree with you, but at the same time I do see it for nodes I'm not directly connected to. Maybe that is its own bug 🤷‍♂️. Here's an example from the mesh map tab, where you can [...]

Even considering all of that, I can still see the signal strength of the device I am connected to directly through Bluetooth, so it is still confusing how I cannot see my own node unless it's reporting signal strength based on what another note reported back if that makes sense.

[image]

[image]