meshtastic/Meshtastic-Apple #891: 🐞 [Bug]: node filters reset moving off node screen

Suhta, 2024-08-21:
### Firmware Version

2.4.0

### What did you do?

On node screen, by default it shows all nodes detected by Lora and MQTT.  However, the filters by default are on for Lora and online, which isn’t being shown.  Changing  the filters to only show Lora and online only works, but if you go off node screen and say to the Bluetooth screen and back to node screen, the filters are not remembered and again reset.  

### Expected Behavior

The filters set should be remembered next time you go to node screen. 

### Current Behavior

See above

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-08-22:
the node filters are not stored anywhere so they don't stick if you close the app but they stay selected during a single app session.

Suhta, 2024-08-23:
They definitely do not stay the same within my single session. I can go from nodes to Bluetooth and back and the filters are changed or at least what is displayed.

bswinnerton, 2024-09-07:
Ideally they would be stored somewhere across sessions.

closed 2024-10-20 by garthvh