meshtastic/Meshtastic-Apple #920: 🚀 [Feature Request]: allow users to hide node "possible locations" ranges on MeshMap

powersjcb, 2024-09-08:
### OS

iOS

### Description

**Problem**

There's a handful of users in my area that have very large position uncertainty configured. These are causing 10+ mi circles to appear on the map for each of them.

**Proposed Solution**

* Add a new MeshMap setting to disable range circles
* Add a new position uncertainty value to the node detail view on MeshMap.

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

Perhaps in the future, deep linking to a node could launch the map UI and center on that node. With the node selected, it enables the uncertainty distance circle for just that node and sets the minimum zoom distance to be ~2-3x the size of that circle. (so the user is made aware if the position is not accurate)

powersjcb, 2024-09-08:
[image]

powersjcb, 2024-09-08:
wow, someone on reddit separately posted a question about what these circles are about a week ago.

They included a nearly nearly identical screenshot of the issue we are seeing on the SLC area map. 

https://www.reddit.com/r/meshtastic/comments/1f5xit0/what_do_the_rings_mean_on_the_meshtastic_app_map/

powersjcb, 2024-09-11:
splitting out this ticket to help users understand these circles when they are shown. https://github.com/meshtastic/Meshtastic-Apple/issues/924

garthvh, 2024-10-20:
Lines seems great, hiding imprecise locations does not match the direction the project is going.

closed 2024-10-20 by garthvh