meshtastic/Meshtastic-Apple #930: 🐞 [Bug]: nodes UIs show “signal good” when RSSI value is unknown (indicated by 0)

powersjcb, 2024-09-14:
### Firmware Version

2.5.*

### What did you do?

View nodes list page

### Expected Behavior

It should show unknown or some other indicator if the data is missing. Ideally hiding the value when zero.

### Current Behavior

UI shows 0 and “good” signal when only SNR is present. 

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

I don’t have a lot of context on this could use help defining more clear behavior when data is missing.

garthvh, 2024-09-14:
Rssi is included in the snr and only tells us if there  is noise, not sure what this would do

powersjcb, 2024-09-14:
I think something might still be wrong in how we display this. I'm pretty sure our code assumes RSSI is always present, but a value of `0` represents. ("not available") Using RSSI of 0 would be extremely an strong signal.

This RSSI of `0` (actually `null`) seems to be getting plugged into some calculation for how we display link quality. ("good" "bad" "fair")

See these two nodes:
```
node 1: {
  rssi: 0 // actually an unknown value
  snr: -15
  display: "good"
node 2:
  rssi: -121
  snr: -10.75
  display: "bad"
``` 

The second node actually has a much SNR, but it's still showing up as a worse link than node 1.

[image]
[image]

powersjcb, 2024-09-14:
Yup, it looks like our code assumes that RSSI will always be available. (which isn't the case)

```
func getLoRaSignalStrength(snr: Float, rssi: Int32, preset: ModemPresets) -> LoRaSignalStrength {
	if rssi > -115 && snr > (preset.snrLimit()) {
		return .good
	} else if rssi < -126 && snr < (preset.snrLimit() - 7.5) {
		return .none
	} else if rssi <= -120 || snr <= (preset.snrLimit() - 5.5) {
		return .bad
	} else { return .fair }
}
```

powersjcb, 2024-09-14:
Per the discord:
> RSSI is not stored in NodeInfo, only SNR. So, if you haven't received any data from a node after connecting, you'll only see the SNR.

GUVWAF, 2025-01-25:
I believe on the NodeInfo it still displays RSSI even if it's unknown (0) because it only checks for the SNR value:
https://github.com/meshtastic/Meshtastic-Apple/blob/4a7f5a261fc2762324623b7f7e6e238b48dda261/Meshtastic/Views/Nodes/Helpers/NodeInfo.swift#L37

Also minor thing: the unit for RSSI is dBm, not dB like for SNR.

closed 2025-04-04 by github-actions[bot]