meshtastic/Meshtastic-Apple #935: 🐞 [Bug]: Navigation glitch in the latest desktop UI

PhilRosa-Leeke, 2024-09-16:
### Firmware Version

2.5.4 (1239)

### What did you do?

The new UI keeps returning to the "Bluetooth" window when ever I click on a "sub" option in the Settings and Messages windows.

### Expected Behavior

To stay within the section and not keep returning me to the Bluetooth window/section.

### Current Behavior

The new UI keeps returning to the "Bluetooth" window when ever I click on a "sub" option in the Settings and Messages windows.

### Participation

- [X] I am willing to submit a pull request for this issue.

PhilRosa-Leeke, 2024-09-16:
Who said anything about iOS?

I am using macOS 15.0 on my M1 Macbook Air.

garthvh, 2024-09-16:
Mac OS 14

PhilRosa-Leeke, 2024-09-16:
As mentioned above, macOS 15 (Sequoia) is what I am using!