meshtastic/Meshtastic-Apple #943: 🚀 [Feature Request]: Disable continual notification sound for Live Activity

akrabu, 2024-09-23:
### OS

iOS

### Description

In a nutshell, when using the Live Activity lock screen widget for iOS, every single time it's updated, I get a notification sound.  It used to be every 60 seconds, and is now every 15 minutes which is better... but I'd prefer if there was no sound at all.  I love the live widget, but I don't need my phone audibly clamoring for my attention more than it already does haha

Looking at the code... I see in Meshtastic/Helpers/LocalNotificationManager.swift on line 58 it says `content.sound = .default` - and googling around, it looks as though setting that to `content.sound = nil` might mute it?  I've also seen "notificationContent.sound = nil" mentioned - but I am not a Swift programmer, just trying to be as helpful as possible!

It would be nice if this were either muted by default, or set to an app setting toggle.  Just a thought.  Thanks!  :)

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-09-23:
Newer firmware is 15 minutes, the notification can't be muted

closed 2024-09-23 by garthvh

akrabu, 2024-09-23:
@garthvh Is there a reason it cannot be muted?  I use many apps with live activity widgets that don't continually hound me with notification sounds...