meshtastic/Meshtastic-Apple #947: 🚀 [Feature Request]: Standalone Chat-Client

mertensm72, 2024-09-26:
### OS

iOS

### Description

I would like to have a chat-only client. So that I could set up nodes an this app for use by older people/non technicians...

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

The iOS client is great, but the navigation is confusing for non technicians. Perhaps an administered node could lead to the client app not showing configurational menus?

powersjcb, 2024-09-28:
[One thing we have already raised](https://github.com/meshtastic/web/issues/282) is that we need a "Getting started with Meshtastic" tutorial in the apps.

@mertensm72 could you describe this in more detail of what exact workflow you have been having with the app?

* What pre-setup have you done for the nodes?
* How are you getting the nodes to them? (in person or mail)
* How is the app configured the first time? (Are you doing it for them, or are they self-serve?)

I'm fairly doubtful we will be crippling the ability to change settings by default, but we definitely want to make this easier. I'd love to see us have a first timer "Setup Wizard" that prompts with something like this.

```
The following best describes my use of Meshtastic
* "I'm new to Meshtastic and need to setup my first node"
* "Someone has setup this node for me, they told me to pair a device so we can chat"
* "I'm a Meshtastic pro"
```

mertensm72, 2024-09-30:
Hi powersjcb, I did not have any probs with the workflow of the app. For me it's fine!

- I set up my nodes via Web und iOS App.
- I would like to set up nodes by my self.
- I could configure the app for the first time.
- After this point, the app should only show Chats. And perhaps the BT state... (See attached mockup)

So the solution an be a stripped down Chat Client App with no configuration functions.

powersjcb, 2024-09-30:
I think it could make sense to merge: BLE devices + chat selection into the same initial screen. (they both have small number of items for most users and it puts more useful information on the landing page after pairing a device)

The nodes list being below the chat options can lead to awkward navigation issues. This makes it so that if you were scrolling down to find a node, find it, now scroll all the to the top to reach Channel/Messages Buttons.

mertensm72, 2024-10-02:
That leads in the right direction. Perhaps a switch in the app settings could solve my vision from fading not used menus from the app itself?
iOS
Settings
Apps
Meshtastic
Administration Navigation on/off
-> 
Complete App as it is now
or
App reduced to Chat functionality

garthvh, 2024-10-20:
You can use the managed mode to remove configuration and manage the node over admin/

closed 2024-10-20 by garthvh

mertensm72, 2024-10-20:
Yes, but the App complexity stays the Same. No sleak top navigation layer... confusing while Not working