meshtastic/Meshtastic-Apple #948: 🐞 [Bug]: Configure remote administration

mertensm72, 2024-09-26:
### Firmware Version

2.5.2

### What did you do?

I want to remotely administer another Node.

### Expected Behavior

Going to Setting and switch from local node to remote node (CONFIGURE). Afterwards: administer remote node

### Current Behavior

CONFIGURE under Settings is not switchable: I can't select the remote node

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

After setting up an admin channel on the 1st node (not on the remote node!), I can switch to remote note and administer it, as expected. admin-channel administration is not activated on both nodes.

mertensm72, 2024-09-26:
iOS 18.0
Meshtastic 2.5.7 (1255)

garthvh, 2024-10-20:
You have to enable administration under app settings

closed 2024-10-20 by garthvh

mertensm72, 2024-10-20:
On which Node? Aktivating it on the nodes to administer it the chooser on the administrative node stays inaktive.
Activating it on the administrative node leads to a no longer adminstratable admin node.