meshtastic/Meshtastic-Apple #949: 🐞 [Bug]: Repeated notifications "Position can only be sent once every 5 seconds"

mosh1, 2024-09-26:
### Firmware Version

2.5.2

### What did you do?

Updated to firmware 2.5.2 and Meshtastic App 2.5.7 (1255). Running on Rak Miswesh. I get repeated notification alerts "Position can only be sent once every 5 seconds". Position setting in Apple Settings is set to send once every 5 minutes.

### Expected Behavior

Should not be getting those notifications if position is set to once every 5 minutes. This is a regression since previous app version and firmware did not have these notifications.

### Current Behavior

I get repeated notification alerts "Position can only be sent once every 5 seconds" when position setting in Apple Settings is set to send once every 5 minutes.

### Participation

- [ ] I am willing to submit a pull request for this issue.

quimnut, 2024-09-26:
T1000-E

I had 2.5.2 on for some time prior to this commencing so it is possible I may have received an ios app update.

Tried turning things off granularly, I found turning channels and smart location off, and force restart the ios client seemed to work but eventually I had to turn location access off via the phone app permissions.

I have the device gps on this morning instead will provide an update if i can re-create this without using the ios app.

quimnut, 2024-09-28:
Not a client issue, reported on android also.

garthvh, 2024-09-29:
Fixed in 2.5.3 firmware

closed 2024-09-29 by garthvh