meshtastic/Meshtastic-Apple #962: 🐞 [Bug]: Traceroutes which pass through nodes on firmware before 2.5 do not return

jonootto, 2024-10-11:
### Firmware Version

2.5.8

### What did you do?

When conducting a traceroute which goes to nodes on firmware prior to 2.5, the traceroute shows in the app as no response, however you can see on apps like meshmap that it has actually gone through. It also works as epxected in CLI.

### Expected Behavior

Traceroute should show.

[image]

### Current Behavior

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

There's been some discussion about this in discord and seems somewhat widespread

jonootto, 2024-10-11:
I actually don't have any longer routes than direct that are >2,5 so its possible this is the case for all traceroutes that are longer than direct.

TyMi, 2024-10-11:
Also here with node on 2.5.6 (re upload) and App version 2.5.8 on iOS 18.0.1. 
I get an error message in Settings -> Logs: 
`Error Updating Core Data
TraceRouteHOp: Error
Domain=NSCocoaErrorDomain
Code=1570 "time is a required value."
UserInfo={NSValidation-
ErrorObject=<TraceRoute-HopEntity: 0x301161770> (entity:
TraceRouteHopEntity; id:
0x3036ebfa0 <x-
coredata:///
TraceRouteHopEntity/ t2663483-767B-4CB3-
A720-4B317DB56EA1112>;
data: <fault>),
NSLocalizedDescription=tim e is a required value., NSValidationErrorKey=time, NSValidationErrorValue=nul
1}`

Direct is working, tested several different other nodes with hops, all the same error.

lysol, 2024-10-12:
Same thing with mine. I had a node I consistently could receive a 1-hop from, but nothing now. I pulled out the SDR and I can see LoRa chirps coming back, the app just doesn't seem to notice.

bjpetit, 2024-10-14:
> Also here with node on 2.5.6 (re upload) and App version 2.5.8 on iOS 18.0.1. I get an error message in Settings -> Logs: `Error Updating Core Data TraceRouteHOp: Error Domain=NSCocoaErrorDomain [...]

This should be addressed by #961

jonootto, 2024-10-20:
Resolved in latest app.

closed 2024-10-20 by jonootto