meshtastic/Meshtastic-Apple #967: 🐞 [Bug]: No sending/receiving possible after changing the device name

Burny08, 2024-10-14:
### Firmware Version

2.5.4

### What did you do?

- Flashed from 2.2.21 to 2.5.5 on a T-Beam V1.2
- After changing the device name with the iOS app, messages no longer come in or out

### Expected Behavior

It would be great if I could change the name of the device and everything would still work afterwards :D

### Current Behavior

Hello everyone, 
I don't know if it's the firmware or the app. I have long since updated my T-Beam from version 2.2.21 to the current 2.5.5. 
The device I use is an iPhone 13 PRO with the latest software. 

Now to my problem: 
I can set the LoRa settings and then everything works well. 
But as soon as I change the name under "User" and set License Operator to on, no messages go out or come in. I have to completely re-flash the T-Beam to be able to use it again. I tested both the current and the oldest stable version. Same problem everywhere. 

PS: I tested with 2x T-Beam. Same problem. I also reinstalled the app on iOS. The web client doesn't work via serial either.

Can anyone help? Am I too stupid or is it actually a bug?
Unfortunately, I can't display a debug on the iPhone, or? 
Unfortunately, I am only a user, not a programmer.

Thank you so much!

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-10-14:
Turning on ham mode deleted your keys.

Burny08, 2024-10-14:
> Turning on ham mode deleted your keys.

Oh, that's that new security tab? It didn't exist back then :D What's that all about? What do I have to do to get everything working again? :) Thank you very much for your help

closed 2024-10-19 by garthvh