meshtastic/Meshtastic-Apple #973: 🚀 [Feature Request]: Allow "Fixed Position" to be manually defined by lat/lon

Xaositek, 2024-10-20:
### OS

iOS, iPadOS, macOS

### Description

If you choose "Fixed Position", the iOS application allows you to define a manual location; one field for latitude, separate for longitude

Looks possible with the Python CLI (meshtastic --setlat 25.2 --setlon -16.8 --setalt 120 / https://meshtastic.org/docs/software/python/cli/usage/) already so the idea seems generally accepted within the community

Additionally with https://github.com/meshtastic/firmware/pull/4713, we should permit Remotely Administered nodes from getting the same benefits.

### Participation

- [X] I am willing to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

garthvh, 2024-10-20:
Perfect CLI functionality

closed 2024-10-20 by garthvh

garthvh, 2024-10-20:
If the position comes from the iOS app I want it to be a real position provided by a GPS, spoofing of position is just not on the iOS roadmap, and has never been.