meshtastic/Meshtastic-Apple #975: 🐞 [Bug]:

jirogit, 2024-10-21:
### Firmware Version

2.5.7.f77c87d

### What did you do?

I go to App Store Meshtastic. I looked at the Version number.

I go to the app setting tab. click "About Meshtastic" > I looked at the Version number

### Expected Behavior

I go to App Store, look at Meshtastic Version, I see  "2.5.9" 

I go to setting tab. click "About Meshtastic" > I see Version: "2.5.9"

### Current Behavior

I go to App Store, look at Meshtastic Version, I see  "2.5.9" 

I go to setting tab. click "About Meshtastic" > I see Version: "2.5.7 (1255)"

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Thank you very very for taking care this app!!

bjpetit, 2024-10-22:
Are you not seeing an update button for the Meshtastic app in the App Store?

https://support.apple.com/en-ng/guide/iphone/iph98709f167/ios

If you are not seeing an update option, what device and os version are you running?

jirogit, 2024-10-22:
It is not problem of App Store.

Meshtastic App for iOS indicates version 2.5.7, even it is actually ver. 2.5.9.

bjpetit, 2024-10-23:
It looks like 2.5.9 requires iOS 17 or later and you're on 16. That's why it's not offering the upgrade.

closed 2024-10-23 by jirogit