meshtastic/Meshtastic-Apple #977: Secondary / Tertiary Admin Keys Connected

AustinSaintAubin, 2024-10-24:
### Firmware Version

2.5.9 (1288) & 2.5.7

### What did you do?

The secondary and tertiary admin keys seem connected in the iOS apps UI. Editing the Secondary Admin Key / Tertiary Admin Key will also edit the Tertiary Admin Key / Secondary Admin Key. They secondary/tertiary keys also do not seem to save, instead assuming the Primary Admin Key when saved and rebooted. Related to #971 

The red box indicating invalid key behaves differently depending on show/hide state.
[image]

### Expected Behavior

Each of the Primary / Secondary / Tertiary admin key fields should operate independently, and should save to their indented settings locations. 
I recommend their default be blank if no saved key value present.

### Current Behavior

They secondary/tertiary keys also do not seem to save, instead assuming the Primary Admin Key when saved and rebooted. Related to #971 

The red box indicating invalid key behaves differently depending on show/hide state.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Side note: the text fields are very small, and difficult to copy or paste a key in to, as pressing outside of the text field(s), which is easy to do, toggle the hide/show function. Could the text fields be made larger, and a separate copy and separate paste button be added to each admin key field?

Xaositek, 2024-11-16:
I believe https://github.com/meshtastic/Meshtastic-Apple/commit/2c4cf4cc57844acd6c916f71344d72021c9b1556 fixed this

AustinSaintAubin, 2024-11-17:
Looks like the admin fields are separated now, however the all 3 of the admin fields do not seem to save OR recall now.

closed 2024-11-27 by garthvh