meshtastic/Meshtastic-Apple #981: 🐞 [Bug]: Inaccurate positioning using IOS in China, resulting in offset issues.

alancccbbb, 2024-10-31:
### Firmware Version

2.5.7 Beta

### What did you do?

The App Store of American accounts installed Meshtastic in China has inaccurate positioning and deviation, with a difference of several hundred meters. According to relevant information, commercial maps in China are not made according to the WGS-84 standard, but based on the GCJ-02 coordinate system; The GCJ-02 coordinate system is encrypted from WGS-84.

### Expected Behavior

The App Store of American accounts installed Meshtastic in China has inaccurate positioning and deviation, with a difference of several hundred meters. According to relevant information, commercial maps in China are not made according to the WGS-84 standard, but based on the GCJ-02 coordinate system; The GCJ-02 coordinate system is encrypted from WGS-84.

### Participation

- [X] I am willing to submit a pull request for this issue.

garthvh, 2024-10-31:
The iOS app just gets a lat long pair, no insight into any of the GPS details.

closed 2024-10-31 by garthvh

garthvh, 2024-10-31:
Also looks like it is intentional https://en.m.wikipedia.org/wiki/Restrictions_on_geographic_data_in_China

hayschan, 2024-10-31:
Hello, @garthvh !

The GPS offset issue with the Meshtastic iOS app, especially when used in China, stems from how geographic data is handled by the Chinese mapping regulations. Below, I’ll outline why this offset occurs and provide a solution to accurately transform coordinates for the app to work in China.

---

### 1. Why the Coordinates Offset Happens in China

The offset issue arises because China mandates unique coordinate systems for any map or GPS data displayed publicly within its borders:
- **WGS-84**: This is the international standard for GPS data, which the Meshtastic hardware outputs. However, using WGS-84 directly on maps in China results in significant position errors.
- **GCJ-02 (Mars Coordinate System)**: This system is used specifically within China, developed by the Chinese State Bureau of Surveying and Mapping to enforce national security policies around geographic data. All GPS-based services in China, including Apple’s MapKit, expect coordinates in GCJ-02. Hence, displaying WGS-84 coordinates on a GCJ-02-based map leads to visible offset errors (typically several hundred meters).
- **BD-09**: An additional layer of encryption over GCJ-02, used only in Baidu Maps.

As Meshtastic hardware outputs WGS-84 coordinates, converting these to GCJ-02 before displaying them on the map in China would resolve the issue. This adjustment is necessary because Apple’s MapKit in China relies on data from Amap (高德地图), which uses the GCJ-02 format, not WGS-84 ([Reference 1](https://www.0daybug.com/posts/d90d8bd/index.html), [Reference 2](https://orochichris.github.io/zhy.github.io/2019/12/13/%E7%B2%BE%E9%80%9AMapKit%E6%98%AF%E4%B8%80%E7%A7%8D%E4%BB%80%E4%B9%88%E6%A0%B7%E7%9A%84%E4%BD%93%E9%AA%8C%EF%BC%9F/)).

### 2. Solution: Implementing a WGS-84 to GCJ-02 Conversion in Meshtastic

To accurately plot coordinates on maps in China, we need to convert WGS-84 to GCJ-02. Here’s an example in Objective-C for use in iOS. This code (coming from this open source [blog](https://www.0daybug.com/posts/d90d8bd/index.html) ) checks if the location is in China and performs the conversion if needed:

```objective-c
#import <Foundation/Foundation.h>
#import <CoreLocation/CoreLocation.h>

@interface XPCoordinateConverter : NSObject

// Convert WGS-84 to GCJ-02
+ (CLLocationCoordinate2D)wgs84ToGcj02:(CLLocationCoordinate2D)coordinate;

@end

@implementation XPCoordinateConverter

#define CC_X_PI (3.14159265358979324 * 3000.0 / 180.0)
#define CC_PI 3.1415926535897932384626
#define CC_A 6378245.0
#define CC_EE 0.00669342162296594323

+ (double)transformLat:(double)x y:(double)y {
    double ret = -100.0 + 2.0 * x + 3.0 * y + 0.2 * y * y + 0.1 * x * y + 0.2 * sqrt(fabs(x));
    ret += (20.0 * sin(6.0 * x * CC_PI) + 20.0 * sin(2.0 * x * CC_PI)) * 2.0 / 3.0;
    ret += (20.0 * sin(y * CC_PI) + 40.0 * sin(y / 3.0 * CC_PI)) * 2.0 / 3.0;
    ret += (160.0 * sin(y / 12.0 * CC_PI) + 320 * sin(y * CC_PI / 30.0)) * 2.0 / 3.0;
    return ret;
}

+ (double)transformLon:(double)x y:(double)y {
    double ret = 300.0 + x + 2.0 * y + 0.1 * x * x + 0.1 * x * y + 0.1 * sqrt(fabs(x));
    ret += (20.0 * sin(6.0 * x * CC_PI) + 20.0 * sin(2.0 * x * CC_PI)) * 2.0 / 3.0;
    ret += (20.0 * sin(x * CC_PI) + 40.0 * sin(x / 3.0 * CC_PI)) * 2.0 / 3.0;
    ret += (150.0 * sin(x / 12.0 * CC_PI) + 300.0 * sin(x / 30.0 * CC_PI)) * 2.0 / 3.0;
    return ret;
}

+ (CLLocationCoordinate2D)wgs84ToGcj02:(CLLocationCoordinate2D)coordinate {
    double lat = coordinate.latitude;
    double lon = coordinate.longitude;
    
    if (![self isLocationInChina:coordinate]) {
        return coordinate; // No conversion needed outside China
    }
    
    double dLat = [self transformLat:lon - 105.0 y:lat - 35.0];
    double dLon = [self transformLon:lon - 105.0 y:lat - 35.0];
    double radLat = lat / 180.0 * CC_PI;
    double magic = sin(radLat);
    magic = 1 - CC_EE * magic * magic;
    double sqrtMagic = sqrt(magic);
    dLat = (dLat * 180.0) / ((CC_A * (1 - CC_EE)) / (magic * sqrtMagic) * CC_PI);
    dLon = (dLon * 180.0) / (CC_A / sqrtMagic * cos(radLat) * CC_PI);
    
    return CLLocationCoordinate2DMake(lat + dLat, lon + dLon);
}

+ (BOOL)isLocationInChina:(CLLocationCoordinate2D)coordinate {
    return coordinate.longitude >= 72.004 && coordinate.longitude <= 137.8347 &&
           coordinate.latitude >= 0.8293 && coordinate.latitude <= 55.8271;
}

@end
```

## Implementation in Meshtastic

1. **Coordinate Detection**: This code includes a function to check if the coordinates are within China's borders.
   
2. **Transformation Functions**: The `transformLat` and `transformLon` functions apply the necessary conversion formulas for accurate placement in the GCJ-02 format.

3. **Integrate in Meshtastic**:
   - Before displaying coordinates on MapKit, use the `wgs84ToGcj02` function to transform GPS coordinates for accurate positioning within China.
   
Example Usage:
   ```objective-c
   CLLocationCoordinate2D originalCoordinate = ... // From Meshtastic hardware
   CLLocationCoordinate2D adjustedCoordinate = [XPCoordinateConverter wgs84ToGcj02:originalCoordinate];
   // Use adjustedCoordinate for display in MapKit
   ```

---

## References

1. [Orochichris’ blog on MapKit’s coordinate system challenges](https://orochichris.github.io/zhy.github.io/2019/12/13/%E7%B2%BE%E9%80%9AMapKit%E6%98%AF%E4%B8%80%E7%A7%8D%E4%BB%80%E4%B9%88%E6%A0%B7%E7%9A%84%E4%BD%93%E9%AA%8C%EF%BC%9F/)
2. [Daybug blog: Practical WGS-84 and GCJ-02 conversions](https://www.0daybug.com/posts/d90d8bd/index.html)

garthvh, 2024-10-31:
There are like 200 downloads ever in china, not really on the roadmap at this time.

hayschan, 2024-10-31:
Thank you for the response, @garthvh! I understand that the user base in China is relatively small, so it may not be a priority for now.

However, would you be open to reviewing and potentially merging a pull request if someone from the community implemented the coordinate conversion fix for Chinese users? The implementation would involve a lightweight conversion of WGS-84 coordinates to GCJ-02, making it compatible with China’s mapping requirements. This approach could allow Chinese users to see accurate positioning without adding significant complexity to the codebase.

If this is something you'd consider, I can provide help facilitate contributions for this specific fix.

Thanks!

garthvh, 2024-10-31:
Does not really seem like a good fit for the iOS app to me, I don't really want to maintain it, or have to deal state regulations in china.