meshtastic/Meshtastic-Apple #987: 🐞 [Bug]: Filter by hops doesn't work as expected

kiwiant, 2024-11-09:
### Firmware Version

2.5.6

### What did you do?

On iPhone 15 Pro:
1. Change to Nodes list
2. Click Filter button (bottom right)
3. Change the Hops away slider from All to Direct then to 1 through to 7. All nodes show for All and Direct but none if hops set to 1 through 7.

I have 55 nodes in my list covering hundreds of kilometres over some big hills so would expect some hops. A traceroute to a node 45km away revealed 3 hops towards it and Direct return however there are also nodes a lot further away.

### Expected Behavior

I would expect to see the list change and show the subset of nodes that are 3 hops when I set the slider to 3 hops. Same for all other options.

### Current Behavior

If the slider is set to All or Direct I see all 55 nodes. Any other setting and the list shows 0 nodes.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

[image]
[image]

garthvh, 2024-11-09:
I don't see any nodes with hops there

garthvh, 2024-11-09:
https://github.com/user-attachments/assets/9c9ebe59-008e-400d-8b83-e8553eecc1f7

Works as expected

closed 2024-11-09 by garthvh

kiwiant, 2024-11-09:
> ScreenRecording_11-08-2024.23-26-01_1.mp4 
> Works as expected

Agree the screen recording shows how I expect it to work. I'm using latest non-beta release of the App so perhaps due to data from the Heltec Lora32 v3 although I find [no reference to the word "hop" in the release notes](https://github.com/meshtastic/firmware/releases) :-(.

kiwiant, 2024-11-10:
> Agree the screen recording shows how I expect it to work. I'm using latest non-beta release of the App so perhaps due to data from the Heltec Lora32 v3 although I find [no reference to the word [...]

For those who stumble across the same issue the solution was to update firmware from 2.5.6 Prod to 2.5.11 Beta. "Hops away" now shows on Nodes list (in place of Signal strength which was showing for all nodes). Nodes list filter also successfully filters list as per the screen recording from @garthvh.