meshtastic/Meshtastic-Apple #992: No Precise Location on Private Primary Channel

foxlaw85, 2024-11-13:
### Firmware Version

2.5.12

### What did you do?

Have private primary and public secondary channel. Private primary is set to: Positions Enabled and Precise Location. GPS Enabled either with node GPS module or with share phone GPS.

### Expected Behavior

Node should share precise location with nodes on private primary, as well as show precise location in the app when connected to said node direct via BLE.

### Current Behavior

Node only shows general location on private primary as well as when connected direct via BLE. Node will correct to precise if fixed location is toggled on, and can be witnessed moving from general to precise via two distinct log points on the node map when fixed location is turned on. It appears the node map, both in the node list and the main node map, is ignoring precise location settings unless fixed position is toggled on.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

[image]

Example of it moving from general to precise when fixed location is turned on. It seems to default to this “general” location each time with a multitude of nodes, including all of my nodes and a couple others locally. Again, think this is a map issue.

garthvh, 2024-11-15:
Those are precise locations, imprecise locations will have a circle showing they area where they location can be contained.

foxlaw85, 2024-11-15:
So the nodes, when they report from that spot or any other spot under non fixed location reporting DO have a green circle around them of about 1.5 miles in width on each side of the node. Only when I set the fixed location does the green circle go away. Attached is an example of another node of mine reporting from the exact same location. [image]

Moosey37, 2024-11-18:
I can confirm the issue. On the Primary Channel (CH 0) there is no option to enable "precise location" as there is on Android. I had to set up precise location using an Android phone before letting the iPhone take charge of my node (TBeam on 2.5.11). Any change of settings on the iPhone on the CH 0 page then breaks the precise location setting and has to be reinstated on the Android phone.

Moosey37, 2024-11-18:
Thank you for the correction & clarification. My bad.
I have just set up a similar configuration to yours so I could get precise location as the iPhone App only seems to allow precise location on a channel that has 128bit or higher encryption.
I now have a Primary Private channel with precise location enabled and it seems to be working as it should do.  So, I am not experiencing the exact same problem.

foxlaw85, 2024-11-19:
I have wiped the app settings and data, factory reset my devices through the app and with the nRF wipe on the web flasher, as well as updated to 2.5.13. No difference. I have tried with 3 different devices now and the same result, so I can rule out device (rak starter kit, wismesh pocket, T echo). I've tried with phone gps on and phone gps off. Same result. Still 1.5 mile circle with 12 sats in view. The gps location updates with enough movement but is imprecise and the green circle remains unless I set to fixed location.

foxlaw85, 2024-11-27:
Interesting development here. I am still having the same issue across all devices (rak starter kit, wismesh packet, T echo). BUT, I received my t1000-e in the mail and after trying to for a day, I AM getting a precise location on this device, with the exact same settings as all other devices. This leads me to believe this may be a firmware issue instead of an iOS app issue.

garthvh, 2024-12-25:
Precise location is not available on any of the known keys.

closed 2024-12-25 by garthvh

armond-avanes, 2025-09-18:
@garthvh Is there any particular reason that this option is available on Android (precise location option on primary public channel) but not on Apple platforms?