meshtastic/Meshtastic-Apple #995: 🚀 [Feature Request]: Alarm Emoji to iOS Critical Alert

Xaositek, 2024-11-16:
### OS

iOS, iPadOS, macOS

### Description

This request is to mimic the "Send Bell" in a "Send Alarm" parameter. This doesn't have a direct ASCII implementation, but there is an option for "Police Car Light" in the Unicode 6.0 specification; it is unicode U+1F6A8.

The intent is that if this character is present in the message it will trigger a Critical Alert (https://developer.apple.com/contact/request/notifications-critical-alerts-entitlement/).

Secondarily the option to trigger a "Send Alarm", would only be on Private Channels, the public channels would display the alarm but would not push a critical alert.

This would also offer additional options for the "External Notification Module" if it permitted additional use cases or pin-out functions, this may require some additional firmware level work for these to trigger and a separate request.

Use Case: We are leveraging Meshtastic in Central Missouri and working now also to build a text variant of our Skywarn presence. The idea is that if a Weather Warning is issued, it can generate a notification that would break through Do Not Disturb and alert users who have opt'd to join our channel.

### Participation

- [X] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

garthvh, 2024-11-23:
This is just an emoji

garthvh, 2024-11-23:
U+1F6A8

Xaositek, 2024-11-23:
Great first step and thank you for taking that initiative! With the concept of not allowing the response on the public channel, we could craft this as an "Opt-In" per channel? Unsure how much room is left in the protobuf for channels so don't want to waste that either if it's a finite resource. 

"Allow 🚨 to trigger iOS Critical Notification?"

Xaositek, 2024-12-13:
This issue has had a bounty set at $1,000 to accomplish.

Anyone who'd like to contribute, please do let me know, every little bit helps!

closed 2025-03-31 by garthvh