meshtastic/Meshtastic-Apple #1001: 🚀 [Feature Request]: Show precision bits in settings and allow more values

oherrala, 2024-11-25:
### OS

iOS

### Description

Currently the app allows setting position precision as four separate values: 12, 5.8, 2.9 and 1.5 km.

Here in Finland (65°N) the latitude and longitude with precision bits are not squares, but tall and skinny rectangles so these given numbers might be misleading. The more north or more south we go the taller and skinnier these rectangles get.

This is how position report with inaccuracy created with precision bits looks when drawn on a map here in Oulu, Finland:

[image]

So for example the setting value of 2.9 km could mean precision bit value 13 if measured from longitude or precision bit value 14 if measured from latitude.

I propose to at least show to precision bit value in UI so community can provide tables for various latitudes, or show the precision bit value and also calculate the bounding box width and height of that setting for current location.

I'd also like to see the setting allow precision bit values up to 16—17.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Additional comments

This is a table I made for Finnish Meshtastic community showing various precision bit values in northern tip and southern tip of Finland.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-11-25:
These values are integers and are set by the firmware, more values are available if you are using a private channel.

closed 2024-11-25 by garthvh

garthvh, 2024-11-27:
The point is for the values to not be precise, so if it is not perfect it is fine.