meshtastic/Meshtastic-Apple #1008: 🐞 [Bug]: invalid config

cracky22, 2024-11-28:
### Firmware Version

i.d.k

### What did you do?

```
Process, Category, Level, Message, Zeitstempel
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.debugLogs)), 27.11.24 21:11
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:11
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:11
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:11
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:11
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:11
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, 🗄️ Data, ℹ️ Info, 💾 [TelemetryEntity] of type <private> Saved for Node: <private>, 27.11.24 21:10
Meshtastic, 📊 Stats, ℹ️ Info, 📈 [Mesh Statistics] Channel Utilization: 0.000000 Airtime: 0.624306 for Node: !da552620, 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🗞️ [BLE] (Notify) characteristic value will be read next, 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 21:10
Meshtastic, 🕸️ Mesh, ℹ️ Info, empty telemetry module config, 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.telemetry)), 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:10
Meshtastic, 🕸️ Mesh, ℹ️ Info, empty store & forward module config, 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.storeAndForward)), 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:10
Meshtastic, 🕸️ Mesh, ℹ️ Info, empty pax counter module config, 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.paxCounter)), 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, ForEach<SubviewsCollection, ID, ModifiedContent<ModifiedContent<ModifiedContent<ModifiedContent<ModifiedContent<PickerItemView<Subview>, _EnvironmentKeyTransformModifier<Bool>>, _EnvironmentKeyWritingModifier<Optional<KeyboardShortcut>>>, ViewInputFlagModifier<HasKeyboardShortcut>>, _TraitWritingModifier<KeyboardShortcutPickerOptionTraitKey>>, _EnvironmentKeyTransformModifier<SymbolVariants>>>: the ID ID(base: AnyHashable("msh/EU_868/Bayern/poing")) occurs multiple times within the collection, this will give undefined results!, 27.11.24 21:10
Meshtastic, Invalid Configuration, 💥 Fault, ForEach<Array<String>, String, Text>: the ID msh/EU_868/Bayern/poing occurs multiple times within the collection, this will give undefined results!, 27.11.24 21:10
Meshtastic, 🕸️ Mesh, ℹ️ Info, empty mqtt module config, 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.mqtt)), 27.11.24 21:10
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:10
Meshtastic, 🕸️ Mesh, ℹ️ Info, empty external notificaiton module config, 27.11.24 21:09
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: Optional(Meshtastic.SettingsNavigationState.externalNotification)), 27.11.24 21:09
Meshtastic, 🍏 Services, ℹ️ Info, 🛣 Routed to NavigationState(selectedTab: Meshtastic.NavigationState.Tab.settings, messages: nil, nodeListSelectedNodeNum: nil, map: nil, settings: nil), 27.11.24 21:09
Meshtastic, 🗄️ Data, ℹ️ Info, 💾 [TelemetryEntity] of type <private> Saved for Node: <private>, 27.11.24 21:09
Meshtastic, 📊 Stats, ℹ️ Info, 📈 [Mesh Statistics] Channel Utilization: 0.000000 Airtime: 0.674972 for Node: !da552620, 27.11.24 21:09
Meshtastic, 🍏 Services, ℹ️ Info, 🗞️ [BLE] (Notify) characteristic value will be read next, 27.11.24 21:09
[... cut ...]
Meshtastic, Invalid Configuration, 💥 Fault, Picker: the selection "800000.0" is invalid and does not have an associated tag, this will give undefined results., 27.11.24 20:53
Meshtastic, Invalid Configuration, 💥 Fault, Bound preference MapScopeRegistryKey tried to update multiple times per frame., 27.11.24 20:51
```

### Expected Behavior

no invalid config

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-14:
The logging for admin config sections is updated in 2.5.11

closed 2024-12-14 by garthvh

krztoff, 2025-02-21:
I am seeing this still in 2.5.18

ralewis85, 2025-09-09:
I ran into this `Picker: the selection "0" is invalid` issue by going into Radio Settings -> Channels -> and attempting to give a name to Channel 0.

By default 0 is named "Primary Channel" with name set to blank.  I tried setting it to "Public" without seemingly any errors.  I then started getting the picket error in my logs and ceased to see any other nodes in my area until I reverted the channel name.