meshtastic/Meshtastic-Apple #1009: 🚀 [Feature Request]: dropdown for timezone

meekowai, 2024-11-29:
### OS

iOS

### Description

Can 'timezone' field be made dropdown with city names like in iOS or windows?
It would be easier than having to type string from [here](https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv)
Thanks. 

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-11-29:
The string is built automatically based on your location

meekowai, 2024-11-29:
Mine wasn’t for some reason and have some folks in my mesh with the same problem-time skew -1hr. They have also bad time zone. Having plain names instead of strings would be easier in case installer fails to recognize location.

garthvh, 2024-11-29:
> Mine wasn’t for some reason and have some folks in my mesh with the same problem-time skew -1hr. They have also bad time zone. Having plain names instead of strings would be easier in case installer fails to recognize location.

If you go under device settings the timezone is automatically populated, all you have to do is save.

garthvh, 2024-11-29:
It is a ridiculously long list, not suitable for a dropdown

meekowai, 2024-11-29:
Maybe it can be translated then into gmt+/-hrs, I mean we need to type it in, still easier than the list and long strings. Everyone gets GMT convention. Thanks Garth.

closed 2024-12-03 by garthvh