meshtastic/Meshtastic-Apple #1012: 🐞 [Bug]: Unable to type-check this expression

norrinr, 2024-12-04:
### Firmware Version

N/A

### What did you do?

Using XCode 15.4, cloned Github project, tried both '2.5.11' and 'main' branches. Build gave same error. This may be something I'm missing in my setup or something I'm doing wrong.

### Expected Behavior

The project should build.

### Current Behavior

MQTTConfig.swift:39:22 The compiler is unable to type-check this expression in reasonable time; try breaking up the expression into distinct sub-expressions

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-04:
You need Xcode 16 now

closed 2024-12-04 by garthvh

norrinr, 2024-12-04:
Ok, thanks. The main readme still says 15.4.

garthvh, 2024-12-04:
Updated to say latest release version