meshtastic/Meshtastic-Apple #1018: 🚀 [Feature Request]: Add msh/EU_{433,868}/$country_code to root topic suggestions

stphnrdmr, 2024-12-14:
### OS

iOS

### Description

It would be really helpful to suggest the country-wide channel to users in addition to the hyper-local ones like the city or town they’re in.

This would help with keeping node info and messages more local, prevent overloading clients when they enable down/uplink on the default channel and help organize private channels into the correct, smaller topics. 

In Germany the suggestions should include msh/EU_{433,868}/DE, or /FR in France and so on. 

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [X] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-14:
This should already in place for regions that are not already a country, do you have location access turned on the reverse geoencoder uses the phone location?  It is a little hard for me to test without being in Europe, can you post a screenshot?

stphnrdmr, 2024-12-15:
Thank you, I just checked again in Germany (Hamburg and Braunschweig) and /DE is the first suggestion. Seems like this can be closed.

closed 2024-12-15 by stphnrdmr