meshtastic/Meshtastic-Apple #1019: 🚀 [Feature Request]: Make it easier to switch Bluetooth connection

kiwiant, 2024-12-15:
### OS

iOS

### Description

I have several nodes and generally connect to them via Bluetooth but trying to switch to a different node is difficult. If I connect to a node on the Bluetooth tab the other nodes disappear. If I then want to connect to a different node the connection seems to hold even if the app is restarted right next to the new node.

If all nodes within Bluetooth range were always listed this would make it very easy to switch between nodes. Another possibility could be to tap the green antenna icon (top right) to disconnect and reveal the node list.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-16:
Once you disconnect from a node (swipe left on the connected node) other nodes that are advertising can be connected to but the data is cleared, the node is the user in meshtastic.

closed 2024-12-16 by garthvh

kiwiant, 2024-12-16:
> Once you disconnect from a node (swipe left on the connected node) other nodes that are advertising can be connected to but the data is cleared, the node is the user in meshtastic.

Thanks. I still get caught by this UI element on the iPhone.

garthvh, 2024-12-16:
> > Once you disconnect from a node (swipe left on the connected node) other nodes that are advertising can be connected to but the data is cleared, the node is the user in meshtastic.
> 
> Thanks. I still get caught by this UI element on the iPhone.

There is also a documentation tip right on the view.

[image]