meshtastic/Meshtastic-Apple #1020: 🐞 [Bug]: Reset NodeDB confirm text is too similar to the Factory Reset confirm text

Shrinks99, 2024-12-16:
### Firmware Version

2.5.15

### What did you do?

In the Device Config settings page at the bottom, the user is presented with two deletion options which (rightfully) both have confirmation [action sheets](https://developer.apple.com/design/human-interface-guidelines/action-sheets).  Tapping each of these options reveals very similar confirmation text, which makes it difficult for users to confirm the exact action they are completing.

### Expected Behavior

`Reset NodeDB` → `Reset and wipe the node database`
`Factory Reset` → `Factory reset your device`

### Current Behavior

`Reset NodeDB` → `Erase all device and app data?`
`Factory Reset` → `Factory reset your device and app`

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-12-21:
The suggested text does not describe what the button does, honestly I think the existing text is more clear.  Not wiping the app node data was considerably more confusing to users which is why both are wiped now.

closed 2024-12-21 by garthvh

Shrinks99, 2024-12-21:
Perhaps this is my misunderstanding then.  Is there a difference between the actions of the two buttons? Maybe this should be a documentation fix instead!

garthvh, 2024-12-21:
I think both the app and docs are fine as is.