meshtastic/Meshtastic-Apple #1038: 🐞 [Bug]: position log overwritten

trutledge215, 2024-12-30:
### Firmware Version

2.5.17

### What did you do?

Smart position on t1000e broadcast 5min  update every 30 sec app setting save data local node 1 remote node 2 connected to phone. 

### Expected Behavior

Log of node 1 position data available in iOS app

### Current Behavior

The node 1 position log gets overridden with every position update 
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Node 1 logging data on the phone, but not node 2 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-12-30:
Are the nodes moving? the iOS app purged redundant positions.

trutledge215, 2024-12-30:
> Are the nodes moving? the iOS app purges redundant positions, and updates the time.

Yes moved about 15 miles and still only the most recent position. I have even done a position exchange and it will appear for a moment then get overridden

b8b8, 2024-12-30:
Private primary channel with high precision set?

trutledge215, 2024-12-30:
> Private primary channel with high precision set?

Correct

b8b8, 2024-12-30:
oh youre using the T1000E. i find that struggles with a GPS update rate more frequent than about 60 seconds. set your 30 to 120 seconds and retry.

trutledge215, 2024-12-30:
I will try. Then comment with an update

closed 2024-12-30 by garthvh

trutledge215, 2024-12-31:
Still occurring after I set gps update to 2 min

trutledge215, 2024-12-31:
For some reason it only appears to be only one of the nodes witch ever one is connected to the phone doesn’t matter

garthvh, 2024-12-31:
There is only a single position stored when the positions are imprecise or are less than 10 meters from the last position https://github.com/meshtastic/Meshtastic-Apple/blob/43884e241bb996ef40ccc9617895b6f6ec588731/Meshtastic/Persistence/UpdateCoreData.swift#L366

trutledge215, 2024-12-31:
How would I fix it I have precise location enabled and I was moving well past the minimum distance

AA7US, 2024-12-31:
I'm also seeing this behavior in the iOS app, which is the only app I use.  I went for a 2-1/2 mile hike today with a RAK Wireless Wismesh Pocket.  Smart Position was enabled, minimum distance was set to 100 meters, GPS update was set to one minute.  I watched the screen on the Wismesh Pocket and could see the position was being updated as expected.  The iOS app only ever showed a single line in the position log during the entire hike, but the position was being updated.  I also saved (downloaded) the position log after the hike and it only contained a single line... the last position update.

AA7US, 2024-12-31:
> There is only a single position stored when the positions are imprecise or are less than 10 meters from the last position

Hi Garth, Does this mean we need to completely disable the "Dilution of precision" setting to get a log of all positions?

[image]

garthvh, 2024-12-31:
The most likely cause is that you are using the default channel with a know key where precise location is not allowed

garthvh, 2024-12-31:
> > There is only a single position stored when the positions are imprecise or are less than 10 meters from the last position
> 
> Hi Garth, Does this mean we need to completely disable the "Dilution of precision" setting to get a log of all positions?
> 
> ![IMG_4931](https://private-user-images.githubusercontent.com/41341300/399405826-1cae1e15-f221-4494-9974-7eabae2ad2a9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2MTg2NzgsIm5iZiI6MTczNTYxODM3OCwicGF0aCI6Ii80MTM0MTMwMC8zOTk0MDU4MjYtMWNhZTFlMTUtZjIyMS00NDk0LTk5NzQtN2VhYmFlMmFkMmE5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMxVDA0MTI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4MTRiYjljOGM0MjkzMjgzYzkxMTEzNzQ3MGEzYzAxNWRlMGYxMTYyZTJhNDc0MWVhYmViNzI4MjM1MTFlZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VFZbN5jx8I3WzTgUrJdAUQRzfJ1fD-hgQNnTuaQ0YYw)

No

trutledge215, 2024-12-31:
I’m operating primary channel on an encrypted with precise enabled

garthvh, 2024-12-31:
You can enable the debug logs under security and see all of the packets in question here, the position log code linked above is very simple and there is not much to go wrong.  In almost every case if only one position is in the log an imprecise location is being sent by the firmware 
[image]

trutledge215, 2025-01-01:
If it is an imprecise location why whiff it only be happening to one

garthvh, 2025-01-02:
> If it is an imprecise location why whiff it only be happening to one

Really no other way to get just one position log record, only other possibility is that your position is less than 15 meters different than the previous one over and over. Those are the only two options here, you can see this in the code linked above.

trutledge215, 2025-01-02:
Looking at the serial debug I see precision is 13 on the bad unit and 32 on the good unit any idea what to do

trutledge215, 2025-01-02:
Got it fixed there is a bug in the iOS app that it will not save/reboot when you edit Chanel settings ie precision.