meshtastic/Meshtastic-Apple #1040: 🐞 [Bug]: iOS/macOS-App is laggy with a lot of messages.

mik-at, 2024-12-30:
### Firmware Version

2.5.14 (1327)

### What did you do?

38C3 in Hamburg happened this weekend and we had I think like between 200 to 300 nodes there. The Primary Channel was very active (100+ messages a day).

### Expected Behavior

The iOS app should work normal.

### Current Behavior

The iOS App got very laggy with viewing and scrolling through the messages. iOS users needed to delete the messages from time to time to get the App responsive again when viewing messages. (same happens on macOS)

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Not sure if I can provide anything useful to help debug the problem - but I though I'm going to open a bug anyway.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2025-01-21 by garthvh

mik-at, 2025-01-21:
Why can't we have chunked loading of messages, where it loads the last 50 or so and loads more of them if I scroll up?

ashleycawley, 2025-07-05:
I have encountered this aswell, I'll copy my msg from Discord:
**iOS App feature idea:** Settings > App Settings > App Data > Auto purge messages older than X days from Primary Channel (predefined drop-down perhaps with choices like 30, 60, 90 days).

This idea comes from a performance issue I encountered on my iPhone 13 Pro Max when using the Meshtastic App, I've noticed the app has got much better at retaining messages for me in the Primary Channel, perhaps months worth of messages in there. The app was getting slower and slower to use in all areas of Meshtastic, I wondered if it could be the chat history slowing it down so I cleared all messages in the Primary channel and it was night and day, sped back up to normal speed, became a lot more responsive.

garthvh, 2026-05-17:
The laggy behavior with 200-300 nodes and 100+ messages/day has been addressed by the same round of performance fixes:

- **PR #1776**: Batch message fetching (eliminates N+1 queries), lazy NavigationLinks, @Query replacing computed properties in message lists.
- **PR #1763** and **PR #1750**: Formatter caching and SwiftData query optimizations.
- **PR #1774**: SessionReplay removal (was accumulating all UI renders in memory).
- **Messages are now capped at 50,000 total** with automatic pruning — so a very active channel (100+ messages/day) won't grow without bound.

The combination of these fixes should make the app usable even at scale events like 38C3.