meshtastic/Meshtastic-Apple #1041: 🐞 [Bug]:

stanhamil, 2025-01-01:
### Firmware Version

2.5.15

### What did you do?

Added the public key of a node into the primary admin section of the iOS app for a remote node (connected via Bluetooth) so as to control the remote node through a secondary device.

### Expected Behavior

To select the remote device from the dropdown list of nodes under “configure” in the iOS app and change/control various settings.

### Current Behavior

The settings aren’t available after selecting the remote node and in orange, a message stating “there has been no response to a request for metadata over the admin channel for this node”.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

All devices share the same private, primary channel.

All keys have been checked and verified as correct.

All nodes can see each other and messages sent as direct messages or on the private, primary channel send and receive with no issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AA7US, 2025-01-01:
@stanhamil Did you follow all of the steps below, to include turning on the "Administration" toggle on both nodes?  I ask because it wasn't mentioned.

This applies to the iOS app.

Node A is the primary radio.  Node B is the remote radio.

1. In the node A menu “Settings → Security”, manually copy and share (i.e.: send) the Public Key from node A to node B.

2. In the node B menu “Settings → Security”, manually paste node A's Public Key into the field labeled “Primary Admin Key”. Click “Save” at the bottom of the screen.

3. On BOTH nodes A and B, turn on the “Administration” toggle found under “Settings → App Settings”.

4. On node A in the “Settings → Configure” menu (which is only seen after turning on “Administration“ in step 1), choose which node you want to change (administer) the settings on… that’s either node A (the default) or node B (the remote node).

5. Still on node A, whichever node that you’ve selected in the “Settings → Configure” menu (previous step) will now be active in all the Settings menus of node A. If you choose node B to configure, you’ll see node B’s name listed on each screen of the Settings menus on node A.

6. After using node A to make changes to remote node B, remember to change back the “Settings → Configure” to node A.  Pay attention to the name of the node being administering on the Settings screens.

stanhamil, 2025-01-01:
All of those steps have been followed, including turning on the Administration toggle in the iOS app for BOTH devices when connected via Bluetooth.When selecting the remote mode to configure, some setting headings are greyed out and even those which aren’t, have empty fields or fields which can’t be changed.    ---- On Wed, 01 Jan 2025 16:07:21 +0000  ***@***.***> wrote ----  
@stanhamil Did you follow all of the steps below, to include turning on the "Administration" toggle on both nodes?  I ask because it wasn't mentioned. 
This applies to the iOS app. 
Node A is the primary radio.  Node B is the remote radio. 
 
 
In the node A menu “Settings → Security”, manually copy and share (i.e.: send) the Public Key from node A to node B. 
 
 
In the node B menu “Settings → Security”, manually paste node A's Public Key into the field labeled “Primary Admin Key”. Click “Save” at the bottom of the screen. 
 
 
On BOTH nodes A and B, turn on the “Administration” toggle found under “Settings → App Settings”. 
 
 
On node A in the “Settings → Configure” menu (which is only seen after turning on “Administration“ in step 1), choose which node you want to change (administer) the settings on… that’s either node A (the default) or node B (the remote node). 
 
 
Still on node A, whichever node that you’ve selected in the “Settings → Configure” menu (previous step) will now be active in all the Settings menus of node A. If you choose node B to configure, you’ll see node B’s name listed on each screen of the Settings menus on node A. 
 
 
After using node A to make changes to remote node B, remember to change back the “Settings → Configure” to node A.  Pay attention to the name of the node being administering on the Settings screens.

AA7US, 2025-01-01:
@stanhamil  Remote administration is working properly for me using the current iOS app, the same version firmware you're using, and following the steps that I posted.

I did notice you mentioned "All devices share the same private, primary channel."  I'm have the default public channel configuration from Meshtastic set as my primary channel.  If you are indeed using a private channel for primary, I wonder if that could be the problem?

garthvh, 2025-01-01:
This works fine, just tested again.

closed 2025-01-01 by garthvh

stanhamil, 2025-01-01:
Garth, is the admin function limited to a specific type of channel set up, or should it work over any channel which is set up on both devices, no matter how it’s set up?