meshtastic/Meshtastic-Apple #1047: 🚀 [Feature Request]: The ability to permanently block nodes

YLSnewb, 2025-01-08:
### OS

iOS, iPadOS, macOS

### Description

Our local Meshtastic net has begun receiving crude / rude / harassing messages from new users.

It would be nice to be able to permanently block these nodes from being able to contact group chats or private messages.  I acknowledge that we can't stop them from receiving messages, and that is not part of the request.

### Participation

- [X] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YLSnewb, 2025-01-08:
For implementation, it would be nice for this to be added to the menu when long-pressing a message, with a brief message with confirmation in case of accidentally selecting the wrong option for the menu.  I recommend "Blocking this user will delete their existing messages from all group and private chats, and prevent their messages from being seen in the future.  They will be able to receive any messages you send, but cannot reply to them."

garthvh, 2025-01-08:
You can long press and ignore any node of your firmware is > 2.5.13

closed 2025-01-08 by garthvh