meshtastic/Meshtastic-Apple #1049: 🚀 [Feature Request]: User Onboarding Flow

garthvh, 2025-01-11:
### OS

iOS, iPadOS, macOS

### Description

Create an onboarding flow in order to walk the users through the assorted OS permissions and firmware privacy and security options.

 

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [X] I am willing to submit a pull request for this feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

danditomaso, 2025-01-11:
Hey @garthvh, it's timely that you created this request, we have another issue in the web repo looking for almost exactly the same thing. I wanted to ask if you'll be the user working on adding this to the IOS app? 

I think it's important we all coordinate this feature so it can be implemented across platforms uniformly (web, ios, android). Let me know your thoughts on this approach, I'm happy to coordinate efforts too

garthvh, 2025-01-11:
Yes

danditomaso, 2025-01-15:
What is your availability this week or next? I'd love to get you, and someone from the Android side together on a voice call so we can figure out what this feature should look like.

jamesarich, 2025-01-17:
for reference - android's current onboarding experience

https://github.com/user-attachments/assets/b2065082-7d3d-4ba9-b501-a94bc8066c2d

garthvh, 2025-01-23:
[image]

[image]

[image]

[image]

garthvh, 2025-01-23:
[image]

[image]