meshtastic/Meshtastic-Apple #1051: 🐞 [Bug]: Some Nodes Over MQTT Cannot Be DM'd by Build 2.5.15 (1337)

Xaositek, 2025-01-13:
### Firmware Version

2.5.18

### What did you do?

I noticed with current version from TestFlight that certain (see images) nodes cannot be DM'd over MQTT despite being only a single hop away while other nodes multiple hops away can be DM'd. This is only with new app builds, older versions actually **do** successfully DM nodes over MQTT so this appears to be an artificial limitation or a limitation that doesn't 100% work/not work as expected.

### Expected Behavior

[image]
[image]
[image]

Here is the CMRG ECHO on iOS version 2.5.14 (1327) - successfully DM'd over MQTT

### Current Behavior

[image]
[image]
[image]
[image]
[image]
[image]

Here is the same node with the same conditions with version 2.5.15 (1337) - no direct message available, but you can see that channels do reach it and it responds.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

[image]
[image]

Adding an example of another bot that is more hops away and DM is available which doesn't make sense.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Xaositek, 2025-01-13:
Possibly linked to issue #799 ?

Xaositek, 2025-01-13:
One note also - this is a private broker MQTT server, this is not the public MQTT server

Xaositek, 2025-01-23:
https://github.com/meshtastic/Meshtastic-Apple/commit/a31fda92d19d5ea184b306506c3b65e93a7e4f68

I think this is the commit that changed this.

RevEngOps, 2025-02-13:
In addition to the above commit reference, looks like this may be due to a hard-coded filter that only allows MQTT clients with a hop count of zero to be directly messaged:  https://github.com/meshtastic/Meshtastic-Apple/blob/main/Meshtastic/Views/Messages/UserList.swift#L49.  I've found that a _very_ temporary way around is to go to "App Settings" and tap "Clear App Data".  This essentially wipes out the MQTT flag until the next device info packet is received.  I'm guessing this was implemented to prevent abuse when using public MQTT servers, but it causes problems for those of us using private servers.  Change looks to have been merged into master here:  3a2f2a76e1a18fbdc6981c9e5b1da35b58c4a746.

violarulan, 2025-04-19:
More info:

I have A as my portable node (MQTT proxy disabled), B as stationary MQTT gateway. I cannot use A to DM any node via MQTT.

hayschan, 2025-04-28:
Same problem here for [MeshCN members](https://meshcn.net).

This is affecting our private MQTT server as well, which should not be the intended target at the first place.

It should only target the default public MQTT server only.

violarulan, 2025-05-01:
Suggestion: If it's hard to implement, an option to turn this feature off would be a good solution though.

Whatnoww, 2025-05-19:
This is still causing issues down to this day even on private MQTT servers. https://github.com/meshtastic/Meshtastic-Apple/pull/814/commits/045bd7124af1a58bfa68efadf7fd16abce2951d7

closed 2025-05-27 by garthvh