meshtastic/Meshtastic-Apple #1055: 🐞 [Bug]:  The option to enter the MQTT account&password is missing in version 2.5.16

Fred0410, 2025-01-14:
### Firmware Version

2.5.7

### What did you do?

I have updated the app in Apple App Store,then reconneted my radio,but I found it that MQTT account&password is missing

### Expected Behavior

The MQTT configuration should include the server, username, and password.

### Current Behavior

The MQTT configuration is missing username, and password.This bug is causing an anomaly when I use MQTT.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nerr, 2025-01-14:
[image]

1JI0O, 2025-01-14:
meeting the same problem

joan7770, 2025-01-14:
Also experiencing this issue on iPhone SE (2020) on iOS 18.1.1.

joan7770, 2025-01-14:
I think I may have found where the problem may be: Meshtastic/Views/Settings/Config/Module/MQTTConfig.swift on line 177:

```Swift
if !proxyToClientEnabled && address != "mqtt.meshtastic.org" {
	HStack {
...
```

I think that if check should be checking if `proxyToClientEnabled` true not false. I am not an iOS developer however and don't have a mac new enough that can build this project to test this.

Xaositek, 2025-01-15:
I found it's hidden until you uncheck the Options "Enabled" and "MQTT Client Proxy" and then they show up

joan7770, 2025-01-18:
Seems this issue was fixed in the app update 2.5.17

closed 2025-01-21 by garthvh