meshtastic/Meshtastic-Apple #1056: 🚀 [Feature Request]: Sort options for nodes list

spaceboots23, 2025-01-14:
### OS

iOS

### Description

In the `nodes` view of the app, it would be very useful to be able to sort the list. As a start perhaps by `rssi`, `distance` ,  `last seen` and `hops`. 

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-01-15:
The nodes sort by favorite, last heard and then node name.  Distance is not available for sort as it is a derived field.

closed 2025-01-15 by garthvh

spaceboots23, 2025-01-15:
Right, but without user selection that’s not sorting, it’s just some arbitrarily defined order. Following industry standards and providing the user sort control would be a better overall outcome. Additionally, just because distance is a derived metric doesn’t mean it’s not a useful sort method.

garthvh, 2025-01-15:
You can use the filters to reduce the size of your list, manual sort options are not on the roadmap, if it is a feature you want to sponsor fiscally it can be revisited. The signal strength is only available for directly connected nodes, so filtering down to that much more limited list gives you visibility into those.  The way the sorting and filtering is done using derived fields is not an option.